Sub-IoT/Sub-IoT-Stack #22: Compilation Instructions

drasko, 2015-11-18:
Hello,
can somebody please provide a compliation instructions. I have tried to compile the project:
```
drasko@Lenin:~/dash7/dash7-ap-open-source-stack/stack/build$ cmake ..
-- The C compiler identification is GNU 5.2.1
-- The CXX compiler identification is GNU 5.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at framework/hal/platforms/CMakeLists.txt:39 (MESSAGE):
  There are no supported platforms for the selected toolchain

-- selected platform: 
CMake Error at framework/hal/platforms/CMakeLists.txt:62 (STRING):
  string no output variable specified

CMake Error at framework/hal/platforms/CMakeLists.txt:65 (ADD_SUBDIRECTORY):
  add_subdirectory given source
  "/home/drasko/dash7/dash7-ap-open-source-stack/stack/build/framework/hal/platforms/platform"
  which is not an existing directory.

CMake Error at framework/CMakeLists.txt:120 (TARGET_INCLUDE_DIRECTORIES):
  Cannot specify include directories for target "PLATFORM" which is not built
  by this project.

CMake Error at framework/CMakeLists.txt:130 (TARGET_COMPILE_DEFINITIONS):
  Cannot specify compile definitions for target "PLATFORM" which is not built
  by this project.

-- Configuring incomplete, errors occurred!
See also "/home/drasko/dash7/dash7-ap-open-source-stack/stack/build/CMakeFiles/CMakeOutput.log".
drasko@Lenin:~/dash7/dash7-ap-open-source-stack/stack/build$ 
```

Obviusly, some PLATFORM should be defined somewhere... What are the options that should be provided to `cmake`?

Instructions found here: http://mosaic-lopow.github.io/dash7-ap-open-source-stack/doxygen/md_gettingstarted.html are confusing as they are using GUI. Can somebody please provice command line options.

christophevg, 2015-11-19:
Hi Drasko,

here's a session dump from zero to complete build, using only the command line...

### 1. Get sources and gcc cross-compiler

Depending on your platform, you'll have to download the appropriate gcc for arm. I tend to put it in a folder next to the Dash7 open source stack:

```bash
$ ls
gcc-arm-none-eabi-4_9-2015q3/

$ git clone https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack.git
Cloning into 'dash7-ap-open-source-stack'...
remote: Counting objects: 19880, done.
remote: Total 19880 (delta 0), reused 0 (delta 0), pack-reused 19880
Receiving objects: 100% (19880/19880), 35.88 MiB | 1003.00 KiB/s, done.
Resolving deltas: 100% (12865/12865), done.
Checking connectivity... done.
```

### 2. Create a build folder and run `cmake` to generate the build instructions for the chosen platform:

```bash
$ mkdir build
$ cd build
$ cmake ../dash7-ap-open-source-stack/stack/ \
        -DTOOLCHAIN_DIR=../gcc-arm-none-eabi-4_9-2015q3/ \
        -DCMAKE_TOOLCHAIN_FILE=../dash7-ap-open-source-stack/stack/cmake/toolchains/gcc-arm-embedded.cmake \
        -DPLATFORM=EFM32GG_STK3700 \
        -DPLATFORM_EFM32GG_STK3700_RADIO=cc1101 \
        -DAPP_D7AP_TEST=on
-- Cross-compiling using gcc-arm-embedded toolchain
-- Cross-compiling using gcc-arm-embedded toolchain
-- detected supported platforms: EFM32GG_STK3700EFM32HG_STK3400ifestmatrix_tp1089
-- selected platform: EFM32GG_STK3700
-- The ASM compiler identification is GNU
-- Found assembler: /Users/xtof/Workspace/tmp/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/xtof/Workspace/tmp/build
```

A quick run-down of the different arguments:
* we point cmake to the open source stack implementation: `../dash7-ap-open-source-stack/stack/`
* `TOOLCHAIN_DIR` points to our gcc cross compiler
* `CMAKE_TOOLCHAIN_FILE` points to the cmake configuration for our gcc cross compiler
* `PLATFORM` is one of the supported platforms (see: `dash7-ap-open-source-stack/stack/framework/hal/platforms`).
* `PLATFORM_<insert chosen platform here in capitals>_RADIO` identifies the radio we want to use (see: also `dash7-ap-open-source-stack/stack/framework/hal/chips`).
* turn on compilation of chose app: `APP_<insert chosen application here in capitals>` (see:  `dash7-ap-open-source-stack/stack/apps`).

### 3. Build it!

```bash
$ make
Scanning dependencies of target CHIP_CC1101
[  1%] Building C object framework/hal/platforms/platform/framework/hal/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101.c.obj
[  2%] Building C object framework/hal/platforms/platform/framework/hal/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101_interface.c.obj
[  3%] Building C object framework/hal/platforms/platform/framework/hal/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101_interface_spi.c.obj
[  3%] Built target CHIP_CC1101
Scanning dependencies of target CHIP_EFM32GG
...
[ 97%] Built target d7ap
Scanning dependencies of target d7ap_test.elf
[ 98%] Building C object apps/d7ap_test/CMakeFiles/d7ap_test.elf.dir/d7ap_test.c.obj
[100%] Linking C executable d7ap_test.elf
[100%] Built target d7ap_test.elf
```

I hope this helps you further in exploring the Dash7 open source stack. Good luck and have fun!

glennergeerts, 2015-11-22:
Thanks, i integrated the steps above in the getting started guide on the site. You are right, the commandline instructions are actually easier compared to the GUI. Closing for now, please let us know if there is something wrong/missing.

closed 2015-11-22 by glennergeerts