Sub-IoT/Sub-IoT-Stack #24: Error linking libframework.a

drasko, 2015-12-20:
I am trying to compile the project, but I am hitting the linking error:

```
drasko@Mali:~/dash7-ap-open-source-stack/build$ cmake ../stack/         -DTOOLCHAIN_DIR=../gcc-arm-none-eabi-4_9-2015q3/         -DCMAKE_TOOLCHAIN_FILE=../stack/cmake/toolchains/gcc-arm-embedded.cmake         -DPLATFORM=EFM32GG_STK3700         -DPLATFORM_EFM32GG_STK3700_RADIO=cc1101         -DAPP_D7AP_TEST=on
-- Cross-compiling using gcc-arm-embedded toolchain
-- Cross-compiling using gcc-arm-embedded toolchain
-- detected supported platforms: EFM32HG_STK3400EFM32GG_STK3700matrix_tp1089ifest
-- selected platform: EFM32GG_STK3700
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/drasko/dash7-ap-open-source-stack/build
drasko@Mali:~/dash7-ap-open-source-stack/build$ make
Scanning dependencies of target CHIP_CC1101
[  1%] Building C object framework/hal/platforms/platform/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101.c.obj
[  2%] Building C object framework/hal/platforms/platform/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101_interface.c.obj
[  4%] Building C object framework/hal/platforms/platform/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101_interface_spi.c.obj
[  4%] Built target CHIP_CC1101
Scanning dependencies of target CHIP_EFM32GG
[  5%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/CMSIS/device/src/system_efm32gg.c.obj
[  6%] Building ASM object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/CMSIS/device/src/startup_gcc_efm32gg.s.obj
[  8%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_assert.c.obj
[  9%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_adc.c.obj
[ 10%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_system.c.obj
[ 12%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_cmu.c.obj
[ 13%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_emu.c.obj
[ 14%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_gpio.c.obj
[ 16%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_usart.c.obj
[ 17%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_rtc.c.obj
[ 18%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_dma.c.obj
[ 20%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_int.c.obj
[ 21%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_lcd.c.obj
[ 22%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_timer.c.obj
[ 24%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_i2c.c.obj
[ 25%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_wdog.c.obj
[ 27%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/dmactrl.c.obj
[ 28%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/gpiointerrupt.c.obj
[ 29%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/segmentlcd.c.obj
[ 31%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/cdc.c.obj
[ 32%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/bsp/bsp_bcc.c.obj
[ 33%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/bsp/bsp_stk.c.obj
[ 35%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_adc.c.obj
[ 36%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_mcu.c.obj
[ 37%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_uart.c.obj
[ 39%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_spi.c.obj
[ 40%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_atomic.c.obj
[ 41%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_timer.c.obj
[ 43%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_system.c.obj
[ 44%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_gpio.c.obj
[ 45%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_pins.c.obj
[ 47%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_i2c.c.obj
[ 48%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_watchdog.c.obj
[ 50%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbd.c.obj
[ 51%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbdch9.c.obj
[ 52%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbdep.c.obj
[ 54%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbdint.c.obj
[ 55%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbh.c.obj
[ 56%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbhal.c.obj
[ 58%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbhep.c.obj
[ 59%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbhint.c.obj
[ 60%] Building C object framework/hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbtimer.c.obj
[ 60%] Built target CHIP_EFM32GG
Scanning dependencies of target PLATFORM
[ 62%] Building C object framework/hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_main.c.obj
[ 63%] Building C object framework/hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_leds.c.obj
[ 64%] Building C object framework/hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_userbutton.c.obj
[ 66%] Building C object framework/hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_sensors.c.obj
[ 67%] Building C object framework/hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_debug.c.obj
[ 68%] Building C object framework/hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_lcd.c.obj
[... 15 lines cut ...]
[ 78%] Built target FRAMEWORK_COMPONENT_random
Scanning dependencies of target FRAMEWORK_COMPONENT_fifo
[ 79%] Building C object framework/components/fifo/CMakeFiles/FRAMEWORK_COMPONENT_fifo.dir/fifo.c.obj
[ 79%] Built target FRAMEWORK_COMPONENT_fifo
Scanning dependencies of target FRAMEWORK_COMPONENT_log
[ 81%] Building C object framework/components/log/CMakeFiles/FRAMEWORK_COMPONENT_log.dir/log.c.obj
[ 81%] Built target FRAMEWORK_COMPONENT_log
Scanning dependencies of target FRAMEWORK_COMPONENT_timer
[ 82%] Building C object framework/components/timer/CMakeFiles/FRAMEWORK_COMPONENT_timer.dir/timer.c.obj
[ 82%] Built target FRAMEWORK_COMPONENT_timer
Scanning dependencies of target FRAMEWORK_COMPONENT_crc
[ 83%] Building C object framework/components/crc/CMakeFiles/FRAMEWORK_COMPONENT_crc.dir/crc.c.obj
[ 83%] Built target FRAMEWORK_COMPONENT_crc
Scanning dependencies of target framework
[ 85%] Building C object framework/CMakeFiles/framework.dir/framework_bootstrap.c.obj
Linking C static library libframework.a
Error running link command: No such file or directory
framework/CMakeFiles/framework.dir/build.make:209: recipe for target 'framework/libframework.a' failed
make[2]: *** [framework/libframework.a] Error 2
CMakeFiles/Makefile2:90: recipe for target 'framework/CMakeFiles/framework.dir/all' failed
make[1]: *** [framework/CMakeFiles/framework.dir/all] Error 2
Makefile:72: recipe for target 'all' failed
make: *** [all] Error 2
drasko@Mali:~/dash7-ap-open-source-stack/build$ 
```

maartenweyn, 2015-12-20:
I will give a shot on this one.

It seems it cannot find your linker binary. Is your path set correctly?

Can you execute make with verbose?

Maarten

-- my phone has a tiny keyboard so I kept this message short

drasko, 2015-12-20:
Here is a verbose output:

```
Linking C static library libframework.a
cd /home/drasko/dash7-ap-open-source-stack/build/framework && /usr/bin/cmake -P CMakeFiles/framework.dir/cmake_clean_target.cmake
cd /home/drasko/dash7-ap-open-source-stack/build/framework && /usr/bin/cmake -E cmake_link_script CMakeFiles/framework.dir/link.txt --verbose=1
CMAKE_AR-NOTFOUND cr libframework.a  CMakeFiles/framework.dir/framework_bootstrap.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/CMSIS/device/src/system_efm32gg.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/CMSIS/device/src/startup_gcc_efm32gg.s.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_assert.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_adc.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_system.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_cmu.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_emu.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_gpio.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_usart.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_rtc.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_dma.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_int.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_lcd.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_timer.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_i2c.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/emlib/src/em_wdog.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/dmactrl.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/gpiointerrupt.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/segmentlcd.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/drivers/cdc.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/bsp/bsp_bcc.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/kits/common/bsp/bsp_stk.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_adc.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_mcu.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_uart.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_spi.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_atomic.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_timer.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_system.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_gpio.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_pins.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_i2c.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/efm32gg_watchdog.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbd.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbdch9.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbdep.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbdint.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbh.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbhal.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbhep.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbhint.c.obj hal/platforms/platform/chips/efm32gg/CMakeFiles/CHIP_EFM32GG.dir/usb/src/em_usbtimer.c.obj hal/platforms/platform/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101.c.obj hal/platforms/platform/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101_interface.c.obj hal/platforms/platform/chips/cc1101/CMakeFiles/CHIP_CC1101.dir/cc1101_interface_spi.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_main.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_leds.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_userbutton.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_sensors.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_debug.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_lcd.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_usb_descriptors.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/stk3700_usb.c.obj hal/platforms/platform/CMakeFiles/PLATFORM.dir/libc_overrides.c.obj components/shell/CMakeFiles/FRAMEWORK_COMPONENT_shell.dir/shell.c.obj components/node_globals/CMakeFiles/FRAMEWORK_COMPONENT_node_globals.dir/ng.c.obj components/random/CMakeFiles/FRAMEWORK_COMPONENT_random.dir/random.c.obj components/fifo/CMakeFiles/FRAMEWORK_COMPONENT_fifo.dir/fifo.c.obj components/log/CMakeFiles/FRAMEWORK_COMPONENT_log.dir/log.c.obj components/timer/CMakeFiles/FRAMEWORK_COMPONENT_timer.dir/timer.c.ob [... cut ...]
framework/CMakeFiles/framework.dir/build.make:209: recipe for target 'framework/libframework.a' failed
make[2]: *** [framework/libframework.a] Error 2
make[2]: Leaving directory '/home/drasko/dash7-ap-open-source-stack/build'
CMakeFiles/Makefile2:90: recipe for target 'framework/CMakeFiles/framework.dir/all' failed
make[1]: *** [framework/CMakeFiles/framework.dir/all] Error 2
make[1]: Leaving directory '/home/drasko/dash7-ap-open-source-stack/build'
Makefile:72: recipe for target 'all' failed
make: *** [all] Error 2
drasko@Mali:~/dash7-ap-open-source-stack/build$ 
```

drasko, 2015-12-20:
Hmm... I am looking at `build/CMakeFiles/2.8.12.2/CMakeCCompiler.cmake`, there are parts like:
```
set(CMAKE_C_COMPILER "arm-none-eabi-gcc")
set(CMAKE_C_COMPILER_ARG1 "")
set(CMAKE_C_COMPILER_ID "GNU")
set(CMAKE_C_COMPILER_VERSION "")
set(CMAKE_C_PLATFORM_ID "")

set(CMAKE_AR "CMAKE_AR-NOTFOUND")
set(CMAKE_RANLIB ":")
set(CMAKE_LINKER "CMAKE_LINKER-NOTFOUND")
```

It looks like cross-tool auto-detection did not work for me.

maartenweyn, 2015-12-21:
You can specify your linker path in cmake to solve CMAKE_AR-NOTFOUND cr libframework.a CMakeFiles/framework.dir/framework_bootstrap.c.obj hal/platforms/

-- my phone has a tiny keyboard so I kept this message short

maartenweyn, 2015-12-21:
You can specify the path to ar however i assume you would have to do this for ld etc as well. I 
should not be necessary and i would suggest to first look into why it cmake is not able to detect 
your toolchain..
Your cmake output shows it uses /usr/bin/arm-none-eabi-gcc instead of the version in ~/
dash7-ap-open-source-stack/gcc-arm-none-eabi-4_9-2015q3/ as you specified in your 
cmake arguments. 

Can you make sure the mentioned path is correct please? Did you install /usr/bin/arm-none-
eabi-gcc from a repository package? 

On Monday, December 21, 2015 07:28:25 AM Weyn Maarten wrote:

You can specify your linker path in cmake to solve CMAKE_AR-NOTFOUND cr libframework.a 
CMakeFiles/framework.dir/framework_bootstrap.c.obj hal/platforms/
-- my phone has a tiny keyboard so I kept this message short

-- You received this message because you are subscribed to the Google Groups "dash7-ap-oss" 
group.To unsubscribe from this group and stop receiving emails from it, send an email to dash7-
ap-oss+unsubscribe@googlegroups.com[3].For more options, visit https://groups.google.com/
d/optout[4].

--------
[1] https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/issues/
24#issuecomment-166163015
[2] https://github.com/notifications/beacon/AFYmjTPU6TVdm-Rgo-NQHXnnwNWzsP4hks5pRy9AgaJpZM4G45rH.gif
[3] mailto:dash7-ap-oss+unsubscribe@googlegroups.com
[4] https://groups.google.com/d/optout

maartenweyn, 2016-01-05:
Hi Drasko,

Any progression on this matter?

Maarten

From: Glenn Ergeerts [mailto:glenn.ergeerts@uantwerpen.be]
Sent: Monday, December 21, 2015 20:37
To: dash7-ap-oss@googlegroups.com
Cc: Weyn Maarten <maarten.weyn@uantwerpen.be>; MOSAIC-LoPoW/dash7-ap-open-source-stack <reply@reply.github.com>
Subject: Re: [dash7-ap-open-source-stack] Error linking libframework.a (#24)

You can specify the path to ar however i assume you would have to do this for ld etc as well. I should not be necessary and i would suggest to first look into why it cmake is not able to detect your toolchain..

Your cmake output shows it uses /usr/bin/arm-none-eabi-gcc instead of the version in ~/dash7-ap-open-source-stack/gcc-arm-none-eabi-4_9-2015q3/ as you specified in your cmake arguments.

Can you make sure the mentioned path is correct please? Did you install /usr/bin/arm-none-eabi-gcc from a repository package?

On Monday, December 21, 2015 07:28:25 AM Weyn Maarten wrote:

You can specify your linker path in cmake to solve CMAKE_AR-NOTFOUND cr libframework.a CMakeFiles/framework.dir/framework_bootstrap.c.obj hal/platforms/

-- my phone has a tiny keyboard so I kept this message short

To unsubscribe from this group and stop receiving emails from it, send an email to dash7-ap-oss+unsubscribe@googlegroups.com<mailto:dash7-ap-oss+unsubscribe@googlegroups.com>.
For more options, visit https://groups.google.com/d/optout.

drasko, 2016-01-21:
Yeap, it was a Cmake version that needed upgrade on my old Debian. 

This is what I use now:
```
drasko@Lenin:~/dash7/build$ cmake --version
cmake version 3.4.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
drasko@Lenin:~/dash7/build$ 
```
and it compiles latest master nicely.

We can close this one.

closed 2016-01-21 by drasko

maartenweyn, 2016-01-21:
On Tuesday, January 5, 2016 at 8:03:33 AM UTC+1, Maarten Weyn wrote:
>
> Hi Drasko,
>
>  
>
> Any progression on this matter?
>

Yes, issue closed: 
https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/issues/24

BR,
Drasko