Sub-IoT/Sub-IoT-Stack #25: DASH7 Mode 2

drasko, 2015-12-20:
Is OSS7 stack based on Dash7 Mode 2 specification as described here: http://www.indigresso.com/wiki/doku.php?id=dash7_mode_2:main, http://www.slideshare.net/peburns/dash7-mode-2-summary, and if not what is the difference? Is this MOde 2 now official spec?

maartenweyn, 2015-12-20:
Hi Drasko,

No, the OSS-7 implementeren the dash7 Alliance protocol 1.0 which is the currently protocol supported by the dash7 Alliance.
The mode 2 was the protocol of 3 years ago.

You van download the 1.0 freely from the Alliance website.

My colleague will answer on your other question. Please subscribe to the oss7 mailing list to pose questions. There will be more people replying.

Maarten

-- my phone has a tiny keyboard so I kept this message short

drasko, 2015-12-20:
Hi Maarteen,
thanks, I will re-post on the mailing list here: https://groups.google.com/forum/#!forum/dash7-ap-oss

BR,
Drasko

drasko, 2016-01-21:
We can close this one.

closed 2016-01-21 by drasko