Sub-IoT/Sub-IoT-Stack #29: EZR32 configurable oscillator source for timer

simonhaines, 2016-04-26:
The EZR32LG timer has the LFXO hard-coded as the oscillator source for the RTC. I am bringing up a platform that does not have an external low-frequency crystal. The `startLfxoForRtc()` function in `ezr32lg_timer.c` should use the `HW_USE_LFXO` macro definition in `platform.h` to select the oscillator source for the timer.

Additionally, the `HW_USE_HFXO` macro is used in the `__ezr32lg_mcu_init()` function to select the oscillator source for the MCU. The file `ezr32lg_mcu.c` does not include `platform.h` (nor does any of its includes) and so never sees the `HW_USE_HFXO` definition, and always defaults to the HFRCO. Should the `hal/chips/ezr32/CMakeLists.txt` file include configuration parameters for oscillator selection?

maartenweyn, 2016-04-26:
Hi Simon

Thanks!

You are right! Can you initiate a pull request?

Maarten

simonhaines, 2016-04-26:
Yes, I will add the `HW_USE_LFXO/HFXO` options into `hal/chips/ezr32/CMakeLists.txt` and generate a `hal/chips/ezr32/inc/chip.h` file with the selected results (with the default to define these two macros). Unless you would prefer some other method?

maartenweyn, 2016-04-26:
I will add Glenn to this discussion, he is the architect of the stack.
Glenn your opinion?

M

glennergeerts, 2016-04-26:
There is already a system in place to generate a header containing configuration parameters at the platform level, i think we can use this here as well.
See `hal/platforms/EFM32GG_STK3700/CMakeLists.txt` for an example, specifically the `PLATFORM_PARAM`, `PLATFORM_OPTION` and `PLATFORM_HEADER_DEFINE`
commands. This wil generate a `platform_defs.h` file in you build directory which contains those platform level config options. The chip implementation can then use these defines, based on the platform (board) used for building (ie platform having an external crystal or not).

simonhaines, 2016-04-27:
I assumed you wanted to keep `/hal/chips/` decoupled from `/hal/platforms/` as I couldn't find any existing dependency. By adding the dependency (including `platform.h`), there is no need to create any CMake configuration option--either the platform has a LFXO or it doesn't.

closed 2016-04-27 by simonhaines

glennergeerts, 2016-04-27:
You are correct, it does not need to be configurable on a platform level and can be hardcoded in `platform.h`. Thanks for the pull request!