Sub-IoT/Sub-IoT-Stack #33: Add em_msc.c to CMakeLists.txt of efm32hg

matthiasds, 2016-05-22:
The file em_msc.c is in the happy gecko library folder but not in CMakeLists.txt in stack\framework\hal\chips\efm32hg
By adding "emlib/src/em_msc.c" To the ADD_LIBRARY (${CHIP_LIBRARY_NAME} OBJECT I solved the issue. I don't  know it is excluded for a reason? But em_msc.c is required for the implementation of flash EEPROM emulation.

maartenweyn, 2016-05-22:
Thx was because it was not used in the stack yet.

If you have a working flash implementation of the D7 files stem you can always push it ;-)

matthiasds, 2016-05-22:
For the moment, I didn't change the OSS7 stack itself but rather added the EEPROM emulation to my specific platform. It does work, but for now I use it primarily to save non-volatile sensor calibration values. I suppose a real OSS7 contribution would need a more general approach with HAL for EEPROM access with optional external EEPROM acces or flash EEPROM emulation for the lower layers?

closed 2018-02-08 by glennergeerts