Sub-IoT/Sub-IoT-Stack #46: debug.h uses framework_defs.h which cannot be found

d7ab, 2016-10-01:
Hi,

I'm trying to compile d7ap_test in Simplicity Studio and getting errors relating to missing file framework_defs.h included form debug.h.  I couldn't fine that file anywhere in your repo.  Am I missing something?

Thanks.

maartenweyn, 2016-10-02:
Framework_defs.h is created by cmake based on your cmake options. did you execute cmake?

For more info: http://mosaic-lopow.github.io/dash7-ap-open-source-stack/doxygen/md_gettingstarted.html

Regards

Maarten

d7ab, 2016-10-02:
Thanks.  I'm new to these tools, sorry if my questions sounds silly.  

Do you know how to cmake dependencies in Simplicity Studio as part of a bigger project?

If you can supply Simplicity Studio project where I can just build d7ap_test - would be great.

I'm on MAC OSX and SS has it's internal toolchain.  Not sure how to use/expose it for external making...

glennergeerts, 2016-10-03:
cmake can generate an Eclipse project (which you can also use in Simplicity Studio) for you by specifiying the correct generator using the `-G "Eclipse CDT4 - Unix Makefile"` option. After running cmake with this option you should be able to import the project by pointing to the build dir.
You can use the toolchain which comes with Simplicity Studio by either specifying the path with `-DTOOLCHAIN_DIR=<path>` or adding it to your PATH env var.
The getting started guide shows how to build d7ap_test http://mosaic-lopow.github.io/dash7-ap-open-source-stack/doxygen/md_gettingstarted.html .

With regard to integrating in a bigger Simplicity Studio project: i personally don't have experience with this. What we typically do is use cmake or Makefiles to build the project and call the oss7 cmake scripts from here. You can probably do something similar using cmake and then generate an eclipse project from this, but i'm not sure how well cmake generator for eclipse copes with nested projects.
Alternatively, you can just use the existing build infrastructure to build a library and then include and link this library in your native Simplicity Studio project. Again, i have no experience in using this in Simplicity Studio but i don't see why it shouldn't work.

closed 2018-02-08 by glennergeerts