Sub-IoT/Sub-IoT-Stack #49: Build is broken

drasko, 2017-01-31:
```
...
[ 94%] Building C object apps/gateway/CMakeFiles/gateway.elf.dir/app.c.obj
[ 95%] Building C object apps/gateway/CMakeFiles/gateway.elf.dir/version.c.obj
[ 96%] Linking C executable gateway.elf
../../framework/libframework.a(aes.c.obj): In function `AES128_ECB_encrypt':
/home/drasko/dash7/dash7-ap-open-source-stack/stack/framework/components/aes/aes.c:506: undefined reference to `hw_aes_ecb128'
../../framework/libframework.a(aes.c.obj): In function `AES128_CTR_encrypt':
/home/drasko/dash7/dash7-ap-open-source-stack/stack/framework/components/aes/aes.c:640: undefined reference to `hw_aes_ctr128'
collect2: error: ld returned 1 exit status
apps/gateway/CMakeFiles/gateway.elf.dir/build.make:122: recipe for target 'apps/gateway/gateway.elf' failed
make[3]: *** [apps/gateway/gateway.elf] Error 1
make[3]: Leaving directory '/home/drasko/dash7/dash7-ap-open-source-stack/build/gateway'
CMakeFiles/Makefile2:1311: recipe for target 'apps/gateway/CMakeFiles/gateway.elf.dir/all' failed
make[2]: *** [apps/gateway/CMakeFiles/gateway.elf.dir/all] Error 2
make[2]: Leaving directory '/home/drasko/dash7/dash7-ap-open-source-stack/build/gateway'
Makefile:83: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/drasko/dash7/dash7-ap-open-source-stack/build/gateway'
Makefile:70: recipe for target 'build/gateway/apps/gateway/gateway.elf' failed
make: *** [build/gateway/apps/gateway/gateway.elf] Error 2
drasko@Lenin:~/dash7/dash7-ap-open-source-stack$ 
```

glennergeerts, 2017-02-01:
All build configurations i'm using are building fine. Can you post the output of `cmake -L` in your build dir please?

glennergeerts, 2017-02-01:
apparently the aes.c file was not compiled for efm32gg chips, so if this is your platform this should be fixed in 8261b4e

drasko, 2017-02-01:
Yes - this is the deafult platform in Makefile: https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/blob/master/Makefile#L12 - I just tried to build it and hit the error.

Your commit fixes the error, thanks. I'll close the issue.

closed 2017-02-01 by drasko

glennergeerts, 2017-02-03:
You are correct, this was confusing. I streamlined the docs and buildsystem, everything is now build using cmake only, without the wrapper Makefile (which i didn't use before). Thanks for the issue

commit 48f206cf (2019-02-22): Merge branch '49-modem-interface-should-notify-remote-side-when-restarted' into 'master'

Resolve "Modem interface should notify remote side when restarted"

Closes #49

See merge request aloxy/oss-7!15