Sub-IoT/Sub-IoT-Stack #53: Regression of serial output in commit 86f779e

christophevg, 2017-04-14:
Commit 86f779e seems to introduce behaviour that interferes with "normal" output.

Up to commit c729596, output of `sensor_push` looks like:
```
[005] AC specifier=0 channel=0
[006] recv response @ 98 dB link budget from:
[007] 00 0B 57 00 00 09 12 83
[008] Command completed successfully
```

After commit 86f779e, the output is partly mangled:
```
[001] AC specifier=0 channel=0
[003] 00 0B 57 00 00 09 12 83␀␀k bu(␡␀ 0␡␀ ��␀
[004] Command completed successfully
```

pnunes30, 2017-04-20:
I'm not denying this issue but I can't reproduce this behaviour and I don't see how the mentioned patch can alter the serial output. Could you please indicate whether the shell is enabled in your build? Thanks.

glennergeerts, 2017-04-20:
I will test on wstk6200a platform when I'm back at the office next week

glennergeerts, 2017-04-26:
I've been testing this a bit and can idd reproduce 'garbage' output interleaved in the traces. However this seems already introduced in 31e1a7ed , which is more logical than the mentioned patch since this touches directly on the UART implementation (it introduced DMA for UART TX). I haven't noticed until now since i only  use JLink RTT for logging. However, i don't understand yet why i did not experience problems yet with the ALP interface which does use UART. This has been tested a lot and there the output is as expected. To be continued..  

In the mean time i would suggest to use JLink RTT (set FRAMEWORK_LOG_OUTPUT_ON_RTT) for logging

christophevg, 2017-05-03:
Sorry, my bad. I checked out every commit in reverse chronological order, until the problem disappeared. Something must have gone wrong. If I pickup development again before you can look into it, I'll have a stab at it.

LOorts-Aloxy, 2021-09-28:
There have been large changes to UART and the stack as a whole in the meantime, closing this issue as it's probably not relevant anymore. If you do still encounter this problem, do not hesitate to re-open this issue.

closed 2021-09-28 by LOorts-Aloxy