Sub-IoT/Sub-IoT-Stack #65: timeServer.h: No such file or directory

floufen, 2018-11-22:
Hi,

When compiling lorawan_node app (`--DLORAWAN_NODE=y`), I get : 
```
Scanning dependencies of target lorawan_node.elf
[ 93%] Building C object apps/lorawan_node/CMakeFiles/lorawan_node.elf.dir/lorawan_node.c.obj
/home/riad/dash7-ap-open-source-stack/stack/apps/lorawan_node/lorawan_node.c:23:24: fatal error: timeServer.h: No such file or directory
 #include "timeServer.h"
                        ^
compilation terminated.
make[2]: *** [apps/lorawan_node/CMakeFiles/lorawan_node.elf.dir/build.make:63: apps/lorawan_node/CMakeFiles/lorawan_node.elf.dir/lorawan_node.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1797: apps/lorawan_node/CMakeFiles/lorawan_node.elf.dir/all] Error 2
```

glennergeerts, 2018-11-22:
The LoRaWAN module is not enabled by default, after enabling this with `-DMODULE_LORAWAN=y` this should work

floufen, 2018-11-22:
Thank you ! It works.

commit 8dffe36a (2019-03-27): Merge branch '65-add-spurious-tx-engineering-mode' into 'master'

Resolve "Add Spurious TX engineering mode"

Closes #65

See merge request aloxy/oss-7!24
closed 2018-11-22 by floufen