Sub-IoT/Sub-IoT-Stack #66: assertion fail in configure_next_event()

meishaoming, 2018-12-08:
Hi, 

I encounter a asser fail while doing some test, sensor_pull on B-L072Z-LRWAN1 platform.

Below is the last lines log:

```
[3859] [PHY] START FG scan @ 147479
[3860] [PHY] THR ISR with IRQ 20

[3861] [PHY] RX Packet Length: 64 
[3862] [NONE] Packet queue alloc 0x20001564 slot 0
[3863] 08 29 E9 D2
[3864] [PHY] read 5 bytes, 59 remaining, FLAGS2 0 

[3865] [PHY] THR ISR with IRQ 20

[3866] [PHY] READ FIFO 59
[3867] 1E 11 AE 46 32 30 39 00 2C 00 30 20 01 42 37 34 34 00 3B 00 44 88 C2 00 5D 01 40 00 02 7B B6
[3868] [PHY] RX done (0 dBm)
[3869] [DLL] guard channel
[3870] [DLL] packet received @ 147557 , RSSI = 0
[3871] [NONE] Packet queue mark received 0x20001688, slot 0assertion "((int32_t)fire_delay) > 0" failed: file "/Users/sam/dash7-ap-open-source-stack/stack/framework/components/timer/timer.c", line 301, function: configure_next_event
```

Then i check the source, it is in file `timer.c` function `configure_next_event()`:

```c
#ifndef NDEBUG	    
			//check that we didn't try to schedule a timer in the past
			//normally this shouldn't happen but it IS theoretically possible...
      fire_delay = (next_fire_time - timer_get_counter_value() - timer_info->min_delay_ticks);
			//fire_delay should be in [0,COUNTER_OVERFLOW_INCREASE]. if this is not the case, it is because timer_get_counter() is
			//now larger than next_fire_event, which means we 'missed' the event
			assert(((int32_t)fire_delay) > 0);
#endif 
```

[image]

If the LPTIM counter increase 1 between two pieces of code framed by red line in the picture, the assert will failed. right?

I change the code below and test it no assert failed here again. 

```c
#ifndef NDEBUG	    
			//check that we didn't try to schedule a timer in the past
			//normally this shouldn't happen but it IS theoretically possible...
      fire_delay = (next_fire_time - timer_get_counter_value() - timer_info->min_delay_ticks) + 1;
			//fire_delay should be in [0,COUNTER_OVERFLOW_INCREASE]. if this is not the case, it is because timer_get_counter() is
			//now larger than next_fire_event, which means we 'missed' the event
			assert(((int32_t)fire_delay) > 0);
#endif
```

I don't know if this modification is correct, I hope I can discuss it with you.

meishaoming, 2018-12-08:
sorry, this assert fail appeared again after this modification. 

i am doing more test and debug on it.

meishaoming, 2018-12-09:
add a log, and reproduce the assert fail.

```c
#ifndef NDEBUG
			//check that we didn't try to schedule a timer in the past
			//normally this shouldn't happen but it IS theoretically possible...
            fire_delay = (next_fire_time - timer_get_counter_value() - timer_info->min_delay_ticks) + 1;
			//fire_delay should be in [0,COUNTER_OVERFLOW_INCREASE]. if this is not the case, it is because timer_get_counter() is
			//now larger than next_fire_event, which means we 'missed' the event
            if (((int32_t)fire_delay) <= 0) {
                log_print_stack_string(LOG_STACK_FWK, "fire_delay %u, next_fire_time %u, counter_value %u, min_delay_ticks %u",
                                       fire_delay, next_fire_time, timer_get_counter_value(),timer_info->min_delay_ticks );
            }
			assert(((int32_t)fire_delay) > 0);
#endif
```

log:

> [237097] [PHY] Switching to sleep mode
[237098] [PHY] resuming from sleep mode
[237099] [PHY] sync_word = f498, ch_coding 2
[237100] [PHY] Switching to sleep mode
[237101] [PHY] resuming from sleep mode
[237102] [PHY] sync_word = f498, ch_coding 2
[237103] [PHY] Switching to sleep mode
[237104] [PHY] resuming from sleep mode
[237105] [PHY] sync_word = f498, ch_coding 2
[237106] [PHY] Switching to sleep mode
[237107] [NONE] fire_delay 0, next_fire_time 6575904, counter_value 6575900, min_delay_ticks 5assertion "((int32_t)fire_delay) > 0" failed: file "/Users/sam/dash7-ap-open-source-stack/stack/framework/components/timer/timer.c", line 306, function: configure_next_event

glennergeerts, 2018-12-10:
ok thanks for the detailed information, I see the problem indeed. I did not notice it myself, probably because I'm not using debug builds for longer running tests, so this code is removed by precompiler. The modification you propose looks fine, can you create a PR please?

Related to this: we will probably re-add support for a normal timer next to LPTIMER as well, since for some internal timings the delay of setting the LPTIMER is too big

meishaoming, 2018-12-10:
Thanks for your reply. i make a [PR](https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/pull/69) for it. Please check if this is as your expectation.

glennergeerts, 2018-12-10:
ok, merged. I will close this for now

commit 627ece3f (2019-04-03): Merge branch '66-low-battery-shutdown' into 'master'

Resolve "low battery shutdown"

Closes #66

See merge request aloxy/oss-7!25
closed 2018-12-10 by glennergeerts