Sub-IoT/Sub-IoT-Stack #68: assertion fail in `time_fired()`

meishaoming, 2018-12-08:
Hi, 

i encounter assertion fail in timer_fired(). After a lot test, there are two case:

case 1:

> [68495] [PHY] READ FIFO 59
[68496] 1E 11 AE 46 32 30 39 00 2C 00 30 20 01 42 37 34 34 00 3B 00 44 88 AC 00 5D 01 40 00 02 FA 24
[68497] [PHY] RX done (0 dBm)
[68498] [DLL] guard channel
[68499] [DLL] packet received @ 2354664 , RSSI = 0
[68500] [NONE] Packet queue mark received 0x20001688, slot 0assertion "NG(timers)[NG(next_event)].f != 0x0" failed: file "/Users/sam/dash7-ap-open-source-stack/stack/framework/components/timer/timer.c", line 342, function: timer_fired

case 2:

> [381941] [NONE] packet len 15
[381942] 0F 01 6E 90 28 04 00 32 20 40 00 02 00 00 37
[381943] [DLL] Packet LENGTH 15, TX DURATION 43
[381944] [DLL] Switched to DLL_STATE_CSMA_CA_STARTED
[381945] [DLL] Tca= 415 with Tc 464 and Ttx 43
[381946] [DLL] Adjusted Tca= 394 = 8747425 - 8747404
[381947] [DLL] slot duration: 43 t_offset: 0 csma ca mode: 1
[381948] [DLL] Switched to DLL_STATE_CCA1assertion "NG(next_event) != NO_EVENT" failed: file "/Users/sam/dash7-ap-open-source-stack/stack/framework/components/timer/timer.c", line 348, function: timer_fired

There is only one place to change the `NG(next_event)` and it is in `configure_next_event()`. But `configure_next_event()` is only called in an atomic context. 

So i doubt that sometimes `configure_next_event()` was not called in an atomic context. The only possibility is that it has been opened in some cases caused by atomic nested.

I made a [pull request](https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/pull/67) deal with irq nest counter for stm32 platform. 

i will do some more test on it.

glennergeerts, 2018-12-08:
How can I reproduce this?
Your theory might be correct indeed, but I need to look into it. I merged the PR already since it needed to be done anyway.

meishaoming, 2018-12-08:
here is my test method:

two B-L072Z-LRWAN1 boards, one is `gateway`, the other is `sensor_pull`. use the default master code.

`gateway` keep sending a unicast read request to `sensor_pull` node every 10 second. below is my python code change from [gateway.py](https://github.com/MOSAIC-LoPoW/oss7-thingsboard-gateway/blob/master/gateway.py) : 

```py
# ... some other code

    def query_node(self):
        unicast_uid = "0x46323039002C0030"
        addressee_id = int(unicast_uid, 16)
        print(addressee_id)
        configuration = Configuration(qos=QoS(resp_mod=ResponseMode.RESP_MODE_ANY),
                addressee=Addressee(access_class=0x11, id_type=IdType.UID, id=addressee_id))

        alp_command = Command.create_with_read_file_action(
                file_id = 0x40,
                length = 2,
                interface_type = InterfaceType.D7ASP,
                interface_configuration = configuration)
        print(alp_command)

        self.modem.execute_command_async(alp_command)

    def run(self):
        self.log.info('Started')

        keep_running = True
        while keep_running:
            self.query_node()
            time.sleep(10)

if __name__ == '__main__':
    Gateway().run()
```

then `sensor_pull` receive the request and send reponse back to `gateway`.

after a while (10 minutes or less an hour), i got the assertion fail log from the `sensor_pull`'s JLinkRTTClient terminal.

meishaoming, 2018-12-09:
> How can I reproduce this?
> Your theory might be correct indeed, but I need to look into it. I merged the PR already since it needed to be done anyway.

There is one place i have found could cause atomic nested problem: 

1. `configure_next_event()` was called in a atomic context
2. in `configure_next_event()`, called `sched_post_task_prio()` which called `start_atomic()` and `end_atomic()` in it.
3. after exit from `sched_post_task_prio()`, the irq is enabled, but still run in `configure_next_event()` function.

After this pull request the problem will be solved.

glennergeerts, 2018-12-13:
I revert the irq nesting counter commit in the PR on master, this seems to be causing problems with UART RX interrupt. Until the cause of this is found it's better to revert so gateway works again. TBC

LOorts-Aloxy, 2021-09-28:
There have been major changes concerning the timer, we assume that this is not a problem anymore. If this is incorrect, don't hesitate to open the issue again.

closed 2021-09-28 by LOorts-Aloxy