Sub-IoT/Sub-IoT-Stack #75: Assert in _fs_verify_magic fails on new B_L072Z_LRWAN1 devboard

ijager, 2019-11-08:
I am trying to run the `Sensor Push` example on a new B_L072Z_LRWAN1 board and it fails on the `fs_init` when it is trying to verify a magic number from eeprom. 

The expected magic number is `0xc234`, but I am reading `0x0`.

Does that mean some eeprom initialization went wrong? Or does this code expect some previously prepared eeprom?

The same thing happens when running the `gateway` example. I am following instructions from the [Getting Started Guide](http://mosaic-lopow.github.io/dash7-ap-open-source-stack/docs/building/). However I am flashing the firmware using onboard stlink + openocd + gdb. 

The generated platform_defs.h is as follows

```
//Warning: This file is automatically generated by CMake. Do NOT EDIT this file directly !!
#ifndef __PLATFORM_DEFS_H__
#define __PLATFORM_DEFS_H__
#define PLATFORM "B_L072Z_LRWAN1"
#define PLATFORM_CONSOLE_UART 1
#define PLATFORM_CONSOLE_LOCATION 0
#define PLATFORM_CONSOLE_BAUDRATE 115200
#define PLATFORM_MODEM_INTERFACE_UART 0
#define PLATFORM_MODEM_INTERFACE_BAUDRATE 115200
#define PLATFORM_NUM_LEDS 4
#define PLATFORM_NUM_BUTTONS 1
#define PLATFORM_NUM_DEBUGPINS 2
#define PLATFORM_SX127X_USE_RESET_PIN
#define PLATFORM_SX127X_USE_DIO3_PIN
#define PLATFORM_SX127X_USE_VCC_TXCO
#define PLATFORM_FS_SYSTEMFILES_IN_SEPARATE_LINKER_SECTION
#define USE_STM32L0XX
#define USE_STM32_COMMON
#define USE_SX127X
#define PLATFORM_B_L072Z_LRWAN1
#endif //__PLATFORM_DEFS_H__
```

The backtrace from gdb:

``` console
(gdb) cont
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
__assert_func (file=file@entry=0x801583c "/dash7-ap-open-source-stack/stack/framework/components/fs/fs.c",
    line=line@entry=174, func=func@entry=0x8015714 <__func__.9436> "_fs_verify_magic",
    failedexpr=failedexpr@entry=0x80157f4 "memcmp(expected_magic_number, magic_number, FS_MAGIC_NUMBER_SIZE) == 0")
    at /dash7-ap-open-source-stack/stack/framework/hal/platforms/B_L072Z_LRWAN1/libc_overrides.c:72
72	    __BKPT (0); // break into debugger, when attached
(gdb) bt
#0  __assert_func (file=file@entry=0x801583c "/dash7-ap-open-source-stack/stack/framework/components/fs/fs.c",
    line=line@entry=174, func=func@entry=0x8015714 <__func__.9436> "_fs_verify_magic",
    failedexpr=failedexpr@entry=0x80157f4 "memcmp(expected_magic_number, magic_number, FS_MAGIC_NUMBER_SIZE) == 0")
    at /dash7-ap-open-source-stack/stack/framework/hal/platforms/B_L072Z_LRWAN1/libc_overrides.c:72
#1  0x0800bb18 in _fs_verify_magic (expected_magic_number=expected_magic_number@entry=0x2000048c <incomplete sequence \302>)
    at /dash7-ap-open-source-stack/stack/framework/components/fs/fs.c:174
#2  0x0800bb78 in _fs_init_permanent_systemfiles (permanent_systemfiles=permanent_systemfiles@entry=0x20000510 <d7ap_filesystem>)
    at /dash7-ap-open-source-stack/stack/framework/components/fs/fs.c:99
#3  0x0800bf3e in fs_init (provisioning=0x20000510 <d7ap_filesystem>)
    at /dash7-ap-open-source-stack/stack/framework/components/fs/fs.c:90
#4  0x08001d26 in d7ap_fs_init () at /dash7-ap-open-source-stack/stack/modules/d7ap/d7ap_fs.c:96
#5  0x080019aa in d7ap_init () at /dash7-ap-open-source-stack/stack/modules/d7ap/d7ap.c:46
#6  0x080001b4 in bootstrap () at /dash7-ap-open-source-stack/stack/apps/sensor_push/sensor_push.c:124
#7  0x0800c742 in scheduler_run () at /dash7-ap-open-source-stack/stack/framework/components/scheduler/scheduler.c:369
#8  0x080064e2 in main () at /dash7-ap-open-source-stack/stack/framework/hal/platforms/B_L072Z_LRWAN1/platf_main.c:117
```

glennergeerts-aloxy, 2019-11-08:
The code expects the EEPROM to be flashed correctly, which is done by for example the `make flash-sensor_push` (or `make flash-sensor_push-fs`) target. This assumes JLink however. I suspect that stlink / openocd does not write to the embedded EEPROM of the STM32L. I have no experience with stlink or openocd  but a quick search seems to confirm this. A workaround might be to use `st-flash` to write the EEPROM as described [here](https://www.purplealienplanet.com/node/56).

Otherwise I suggest to reflash the stlink with the JLink firmware as described in the [docs](http://mosaic-lopow.github.io/dash7-ap-open-source-stack/docs/platform-lrwan1/). That way you can use the make targets to write the EEPROM.

ijager, 2019-11-08:
Ah Thanks! I'll try out the workaround and as a last resort convert to JLink.

ijager, 2019-11-11:
So I got it working using the workaround you posted. That post is a bit outdated and the [stlink tool](https://github.com/texane/stlink) became more capable in the meantime, which was nice.

Here are the steps I had to take. Might be useful for some.

There are 2 linker sections that need to go to eeprom:
* .d7ap_fs_metadata_section
* .d7ap_fs_permanent_files_section

We need to copy those to a separate binary file. But first we need to know the starting addresses for those sections:

```console
arm-none-eabi-objdump --headers gateway.elf

gateway.elf:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00015ec8  08000000  08000000  00010000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .ARM.exidx    00000008  08015ec8  08015ec8  00025ec8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .d7ap_fs_metadata_section 00000bfc  08080000  08080000  00040000  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  3 .d7ap_fs_permanent_files_section 0000076c  08080bfc  08080bfc  00040bfc  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  4 .stack        00000500  20000000  20000000  00050000  2**0
                  ALLOC
  5 .data         000000d0  20000500  08015ed0  00030500  2**2
                  CONTENTS, ALLOC, LOAD, DATA
```

The first section (.d7ap_fs_metadata_section) starts at 0x08080000 (=0x0 for eeprom) and has size 0xbfc. So that is where the next section (.d7ap_fs_permanent_files_section) starts. We can now use these numbers to copy that data to a new hex file and then create the binary:

```console
arm-none-eabi-objcopy \
-j .d7ap_fs_metadata_section --set-section-flags=.d7ap_fs_metadata_section="alloc,load" --change-section-lma .d7ap_fs_metadata_section=0 \
-j .d7ap_fs_permanent_files_section --set-section-flags=.d7ap_fs_permanent_files_section="alloc,load" --change-section-lma .d7ap_fs_permanent_files_section=0xbfc \
-O ihex gateway.elf gateway.eep.hex

# Create binary from intel hex
arm-none-eabi-objcopy -I ihex -O binary gateway.eep.hex gateway.eep.bin
```

Finally we can flash the eeprom data using the modified [stlink tool](https://github.com/texane/stlink) as described in the [blogpost](https://www.purplealienplanet.com/node/56). For example using [this fork](https://github.com/manuvelayudhan/stlink/tree/eeprom_write).

```console
./st-flash --reset write build/apps/gateway/gateway.eep.bin 0x08080000

st-flash 1.5.1-50-g3690de9
2019-11-11T08:55:16 INFO common.c: Loading device parameters....
2019-11-11T08:55:16 INFO common.c: Device connected is: L0x Category 5 device, id 0x20086447
2019-11-11T08:55:16 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x1911c00 bytes (25671 KiB) in pages of 128 bytes
2019-11-11T08:55:16 INFO common.c: Ignoring 144 bytes of 0x00 at end of file
2019-11-11T08:55:16 INFO common.c: Attempting to write 4824 (0x12d8) bytes to stm32 address: 134742016 (0x8080000)
Flash page at addr: 0x08081280 erased
2019-11-11T08:55:17 INFO common.c: Finished erasing 38 pages of 128 (0x80) bytes
2019-11-11T08:55:17 INFO common.c: Starting Half page flash write for STM32L core id
2019-11-11T08:55:17 INFO flash_loader.c: Successfully loaded flash loader in sram
 75/75 halfpages written
2019-11-11T08:55:22 INFO common.c: Starting verification of write complete
2019-11-11T08:55:22 INFO common.c: Flash written and verified! jolly good!
```

closed 2019-11-11 by ijager

glennergeerts-aloxy, 2019-11-12:
Good to hear and thanks for the write-up!
As an additional comment: the cmake build already generates an hex file the eeprom sections, for instance `apps/gateway/gateway-eeprom-fs.hex`, so I think you can save a few steps by using this

ijager, 2019-11-12:
I noticed that, but that one turned out to be empty. 

I just did a quick search and I think the section names have been renamed at some point, but not in the cmake file.
in [app_macrs.cmake](https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/blob/2616064bd3a276ae3c2130ac19fca990185008d4/stack/cmake/app_macros.cmake#L158) the section are called `".d7ap_fs_metadata",  ".d7ap_fs_file_headers", ".d7ap_fs_systemfiles"` but in the code and linker file they are `.d7ap_fs_metadata_section,
.d7ap_fs_permanent_files_section`

ijager, 2019-11-12:
Oh and by the way, I figured out that with the [Black Magic Probe](https://github.com/blacksphere/blackmagic) you can just use the original ELF file, and it will automatically flash the code and eeprom correctly. So that's even easier if you have a BMP.

glennergeerts-aloxy, 2019-11-12:
> I just did a quick search and I think the section names have been renamed at some point, but not in the cmake file.

You are completely correct, I seem to have forgotten to change this in cmake and docs.
Thanks for the PR!