Sub-IoT/Sub-IoT-Stack #76: simple_leds app fails to link: undefined reference to `d7ap_volatile_files_data'

ijager, 2019-11-08:
platf.main.c has a reference to [`d7ap_volatile_files_data`](https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/blob/2616064bd3a276ae3c2130ac19fca990185008d4/stack/framework/hal/platforms/B_L072Z_LRWAN1/platf_main.c#L42)

However since this example only does LEDs, `d7ap` module does not get linked. Creating this mismatch which fails at the linker stage.

**Workaround:**

Adding 

```
// The userfiles are only stored in RAM for now
uint8_t d7ap_volatile_files_data[FRAMEWORK_FS_VOLATILE_STORAGE_SIZE];
```

to `simple_leds/app.c` makes it compile and link again. But I am too new to this code base to know the best solution, maybe some preprocessor magic in `platf_main.c`?

glennergeerts-aloxy, 2019-11-08:
Thanks for reporting. I was not aware of this, `simple_leds` is not used anymore and not build by CI so this slipped through. Indeed we assume `d7ap` to be linked for all applications and this would need some refactoring in order to be able to link without this. Do you have a specific use case for not linking `d7ap` ? Otherwise I would suggest to remove `simple_leds` application since it is not used anymore.

ijager, 2019-11-08:
I don't have a use case for not linking d7ap. However, I think a simple blinky example is always very useful to check if you got the toolchain working correctly.

awneil, 2021-03-25:
I also came across this - and came to the same workaround.

I'm with @ijager: For a newbie to the stack, I think this is a valuable example to prove that one can build, download, run, and debug a simple standalone app on a single board - with none of the extra complications of getting radio comms working.

Please don't remove it!

LOorts-Aloxy, 2021-09-28:
this issue should be fixed in [this commit](https://github.com/Sub-IoT/Sub-IoT-Stack/commit/8ff2387dd06cfc2c8b905d66afaae981e212ed57). I just added the variable as suggested. Thank you for flagging this.

closed 2021-09-28 by LOorts-Aloxy