Sub-IoT/Sub-IoT-Stack #85: stm32l0xx hal: support TSC and other peripheral drivers

sjorsdewit, 2020-05-12:
I am working on a project where I use the stm32l0xx TSC peripheral. This specific file appears to be missing  in CMakeLists.txt:

`Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_tsc.c`

Some other files are commented-out as well:

https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/blob/81742d70b00124335af5174a311f864e34077f31/stack/framework/hal/chips/stm32l0xx/CMakeLists.txt#L56-L80

I could make a pull-request to fix this. Or is there a reason that these files were excluded?

glennergeerts, 2020-05-12:
We only compile the STM32 HAL drivers which are needed by the HAL API and to run the actual networking stack here. 
The goal is not to become a general purpose RTOS with a broad HAL API (in fact, we want to move more into the direction of slimming down to a library implementing the networking stack, running on top of another RTOS). 
What you could do here is compile the file in your application level CMakeLists.txt. Does that help you?

sjorsdewit, 2020-05-12:
Ah, ok. That makes sense. I was assuming you were heading in the custom RTOS direction. I'll include it at the application level then, no problem. Thanks for the feedback!

closed 2020-05-12 by sjorsdewit