Sub-IoT/Sub-IoT-Stack #96: Compilation Error while running make

Conaras, 2020-10-21:
Hello! Building of example in ubuntu 20.04 fails. I'm trying to run the make command but it gets stuck at approximately 88 or 90% and the error: "/usr/bin/objcopy: Unable to recognise the format of the input file `gateway.elf'
make[2]: * [apps/gateway/CMakeFiles/gateway.elf.dir/build.make:106: apps/gateway/gateway.elf] Error 1
make[2]: * Deleting file 'apps/gateway/gateway.elf'
make[1]: * [CMakeFiles/Makefile2:1634: apps/gateway/CMakeFiles/gateway.elf.dir/all] Error 2
make: * [Makefile:84: all] Error 2", is printed.

LOorts-Aloxy, 2020-10-21:
Hi, I'm also using Ubuntu 20.04 and am not getting this problem. Did you change any cmake variables and which version of make are you using? Just trying to reproduce the problem on my machine.

Conaras, 2020-10-21:
I am using GNU Make 4.2.1 version and cmake 3.16.3 version and the cmake
command I've typed is the following:
cmake ../dash7-ap-open-source-stack/stack/ -DAPP_GATEWAY=y
-DAPP_SENSOR_PUSH=y -DTOOLCHAIN_DIR=../gcc-arm-none-eabi-9-2020-q2-update
-DAPP_SENSOR_PULL=y -DAPP_SENSOR_ACTION=y

glennergeerts, 2020-10-21:
In the error output it shows it is using "/usr/bin/objcopy" which is the native toolchain, not 
the ARM one (it should be using "arm-none-eabi-objcopy" instead) so something is wrong 
with ARM toolchain detection.

The docs (http://mosaic-lopow.github.io/dash7-ap-open-source-stack/docs/building/[1] ) 
assume that the toolchain is in your PATH, is this the case for you?

Can you paste the full output starting from a clean build dir please?

On Wednesday, October 21, 2020 4:58:11 PM CEST Conaras wrote:

I am using GNU Make 4.2.1 version and cmake 3.16.3 version and the cmakecommand I've 
typed is the following:cmake ../dash7-ap-open-source-stack/stack/ -DAPP_GATEWAY=y-
DAPP_SENSOR_PUSH=y -DTOOLCHAIN_DIR=../gcc-arm-none-eabi-9-2020-q2-update-
DAPP_SENSOR_PULL=y -DAPP_SENSOR_ACTION=y

cmake variables and which version of make are you using? Just> trying to reproduce the 
problem on my machine.>> —>
AMB6L6M4TV2H3SPH3CGBL5DSL3YEVANCNFSM4SZ3IPKQ>> .>

--------
[1] http://mosaic-lopow.github.io/dash7-ap-open-source-stack/docs/building/
[2] https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/issues/
96#issuecomment-713638662
ABLEMIWXNZIBAXMNHR7LMLTSL3ZIHANCNFSM4SZ3IPKQ
[4] https://github.com/notifications/beacon/
ABLEMIV24WRKQ4KJQY7EDQ3SL3ZIHA5CNFSM4SZ3IPK2YY3PNVWWK3TUL52HS4DFVREXG
43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFKEUGBQ.gif

Conaras, 2020-10-23:
I'm sorry for the late response but i was experiencing some technical
issues with my pc.This is the output when i run make in a clean directory
after running  cmake ../dash7-ap-open-source-stack/stack -DAPP_GATEWAY=y
-DAPP_SENSOR_PUSH=y -DAPP_SENSOR_PULL=y
-DTOOLCHAIN_DIR=../gcc-arm-none-eabi-9-2020-q2-update:

Scanning dependencies of target CHIP_SX127X
[ 1%] Building C object
framework/hal/platforms/platform/chips/sx127x/CMakeFiles/CHIP_SX127X.dir/sx127x.c.obj
[ 1%] Built target CHIP_SX127X
Scanning dependencies of target CHIP_STM32L0XX
[ 2%] Building ASM object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/startup/startup_STM32L072xx.s.obj
[ 3%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/stm32l0xx_hal_msp.c.obj
[ 4%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/system_stm32l0xx.c.obj
[ 5%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal.c.obj
[ 6%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_cortex.c.obj
[ 7%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_flash.c.obj
[ 8%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_flash_ex.c.obj
[ 10%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_gpio.c.obj
[ 11%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_i2c.c.obj
[ 12%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_iwdg.c.obj
[ 13%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr.c.obj
[ 14%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr_ex.c.obj
[ 15%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc.c.obj
[ 16%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc_ex.c.obj
[ 17%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc.c.obj
[ 19%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc_ex.c.obj
[ 20%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_spi.c.obj
[ 21%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_tim.c.obj
[ 22%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_tim_ex.c.obj
[ 23%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_lptim.c.obj
[ 24%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_uart.c.obj
[ 25%] Building C object
framework/hal/platforms/platform/chips/stm32l0xx/CMakeFiles/CHIP_STM32L0XX.dir/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_ll_gpio.c.obj
[ 25%] Built target CHIP_STM32L0XX
Scanning dependencies of target CHIP_STM32_COMMON
[ 26%] Building C object
framework/hal/platforms/platform/chips/stm32_common/CMakeFiles/CHIP_STM32_COMMON.dir/stm32_common_atomic.c.obj
[ 28%] Building C object
framework/hal/platforms/platform/chips/stm32_common/CMakeFiles/CHIP_STM32_COMMON.dir/stm32_common_gpio.c.obj
[ 29%] Building C object
framework/hal/platforms/platform/chips/stm32_common/CMakeFiles/CHIP_STM32_COMMON.dir/stm32_common_i2c.c.obj
[ 30%] Building C object
framework/hal/platforms/platform/chips/stm32_common/CMakeFiles/CHIP_STM32_COMMON.dir/stm32_common_mcu.c.obj
[ 31%] Building C object
framework/hal/platforms/platform/chips/stm32_common/CMakeFiles/CHIP_STM32_COMMON.dir/stm32_common_spi.c.obj
[ 32%] Building C object
framework/hal/platforms/platform/chips/stm32_common/CMakeFiles/CHIP_STM32_COMMON.dir/stm32_common_system.c.obj
[ 33%] Building C object
[... 128 lines cut ...]
modules/modules/d7ap/CMakeFiles/d7ap.dir/packet.c.obj
[ 84%] Building C object modules/modules/d7ap/CMakeFiles/d7ap.dir/dll.c.obj
[ 85%] Building C object modules/modules/d7ap/CMakeFiles/d7ap.dir/phy.c.obj
[ 86%] Linking C static library libd7ap.a
[ 86%] Built target d7ap
Scanning dependencies of target d7ap_fs
[ 87%] Building C object
modules/modules/d7ap_fs/CMakeFiles/d7ap_fs.dir/d7ap_fs.c.obj
[ 88%] Building C object
modules/modules/d7ap_fs/CMakeFiles/d7ap_fs.dir/__/__/fs/d7ap_fs_data.c.obj
[ 89%] Linking C static library libd7ap_fs.a
[... cut ...]
[ 91%] Building C object apps/gateway/CMakeFiles/gateway.elf.dir/app.c.obj
[ 92%] Building C object
apps/gateway/CMakeFiles/gateway.elf.dir/version.c.obj
[ 93%] Linking C executable gateway.elf
/usr/bin/objcopy: Unable to recognise the format of the input file
`gateway.elf'
make[2]: * [apps/gateway/CMakeFiles/gateway.elf.dir/build.make:106:
apps/gateway/gateway.elf] Error 1
make[2]: * Deleting file 'apps/gateway/gateway.elf'
make[1]: * [CMakeFiles/Makefile2:1658:
apps/gateway/CMakeFiles/gateway.elf.dir/all] Error 2
make: * [Makefile:84: all] Error 2

closed 2020-10-23 by Conaras