Sub-IoT/Sub-IoT-Stack #101: noise_logger.c compile fails - userbutton.h not found

awneil, 2021-03-25:
```C
[ 83%] Building C object apps/noise_logger/CMakeFiles/noise_logger.elf.dir/noise_logger.c.obj
...stack\apps\noise_logger\noise_logger.c:32:10: fatal error: userbutton.h: No such file or directory
 #include <userbutton.h>
          ^~~~~~~~~~~~~~
compilation terminated.
```

There is, indeed, no such file anywhere in the source tree.

Source tree cloned from GitHib this morning; SHA-1: 42d6bbff887b5a1789f5f01795f3d111467c400c

CMake command line:
`cmake ..\stack\ -DFRAMEWORK_CONSOLE_ENABLED=y -DPLATFORM_USE_NUCLEO_IKS01A2_SHIELD=y -DAPP_GATEWAY=y -DAPP_SENSOR_PUSH=y -DAPP_SENSOR_ACTION=y -DAPP_SENSOR_PULL=y -DAPP_NOISE_LOGGER=y -DAPP_SIMPLE_LEDS=y -DAPP_SNIFFER=y -G "MinGW Makefiles"`

awneil, 2021-03-25:
Also `<platform_sensors.h>` - although this one does exist for the EZR32LG_WSTK6200A and EFM32GG_STK3700 platforms

and other errors ...

glennergeerts, 2021-03-26:
Indeed, we used to have an API for button's before but this was removed. I removed the noise_logger app now as well in c25da89, since it was not used anymore

closed 2021-03-26 by glennergeerts