Sub-IoT/Sub-IoT-Stack #103: I2C on B_L072Z_LRWAN1 platform

awneil, 2021-03-26:
The HTS221 temperature sensor code in the Sensor Push, Pull, and Action examples doesn't work: apart from the initial toggling of SCK, there is no activity on the I2C bus.

This is because the final thing `i2c_init` does is to disable the clock to the I2C controller:
```C

  I2C_CLK_DISABLE((uint32_t)(i2c_ports[idx].i2c));

  return &handle[idx];
}
```
and there is nothing in the rest of the code to re-enable it.

A fix is to add `i2c_acquire(hts221_handle)` and `i2c_release(hts221_handle)` around anything which requires I2C.

Having done that, I do see activity on the bus, and get temperature readings.

Using SHA-1: 42d6bbff887b5a1789f5f01795f3d111467c400c

glennergeerts, 2021-03-26:
Thanks, should be fixed in c25da89. I don't have the sensor shield available so it would be helpful if you could test this..

closed 2021-03-26 by glennergeerts

awneil, 2021-03-26:
With the Sensor Action example from c25da89c09ff2bb11820479fe0cfbc010f196656, I get:
```
00> [000] Device booted
00> 
00> 
00> [001] temp 226 dC
00> 
00> [002] tried to start an interface (226) that is not registered
00> 
00> [003] temp 226 dC
00> 
00> [004] tried to start an interface (226) that is not registered
00> 
```
with a little heat:
```
00> 
00> [053] temp 363 dC
00> 
00> [054] temp 366 dC
00> 
00> [055] temp 373 dC
00> 
00> [056] temp 356 dC
00> 
00> [057] temp 350 dC
00> 
00> [058] temp 347 dC
00> 
```
so values >255 displaying OK.

Not sure the significance of the "tried to start an interface (xxx) that is not registered" - but that behaviour is unchanged