Sub-IoT/Sub-IoT-Stack #105: Sensor Action example: incorrect value in RTT logs

awneil, 2021-03-26:
example:
```
00> [000] Sensor Action: Device booted
00> 
00> 
00> [001] tried to start an interface (231) that is not registered
00> 
00> [002] temp -6400 dC
00> 
00> [003] tried to start an interface (231) that is not registered
00> 
00> [004] temp -6400 dC
00> 
```
It really isn't that cold on my bench!

In `execute_sensor_measurement`:
``` C
  temperature = __builtin_bswap16(temperature); // need to store in big endian in fs
  d7ap_fs_write_file(SENSOR_FILE_ID, 0, (uint8_t*)&temperature, SENSOR_FILE_SIZE, ROOT_AUTH);

  log_print_string("temp %i dC", temperature);
```

The `log_print_string` needs to come before the `__builtin_bswap16` - then we get a sensible value:

```
00> [000] Sensor Action: Device booted
00> 
00> 
00> [001] temp 230 dC
00> 
00> [002] tried to start an interface (230) that is not registered
00> 
00> [003] temp 230 dC
00> 
00> [004] tried to start an interface (230) that is not registered
00> 
00> [005] temp 231 dC
00> 
00> [006] tried to start an interface (231) that is not registered
00> 
00> [007] temp 230 dC
```

SHA-1: 42d6bbff887b5a1789f5f01795f3d111467c400c

glennergeerts, 2021-03-26:
Ok thanks for this, I will push this fix.
You are more then welcome to submit a PR with your fix in the future btw.

glennergeerts, 2021-03-26:
Fixed in c25da89

closed 2021-03-26 by glennergeerts

awneil, 2021-03-26:
Thanks.

Not really a Git user - so someone would have to walk me through the whole "PR" process ...

;)

awneil, 2021-03-26:
Example of fixed output - see #103