Sub-IoT/Sub-IoT-Stack #106: STM32 hw_get_unique_id() bug

ijager, 2021-05-06:
I believe that the way the UID is read on the STM32 platform is buggy.

From [`stm32_common_system.c`](https://github.com/MOSAIC-LoPoW/dash7-ap-open-source-stack/blob/c25da89c09ff2bb11820479fe0cfbc010f196656/stack/framework/hal/chips/stm32_common/stm32_common_system.c#L210):
```c
uint64_t hw_get_unique_id()
{
  return (*((uint64_t *)(UID_BASE + 0x04U)) << 32) + *((uint64_t *)(UID_BASE + 0x14U));
}
```

The manual RM0376 section 33.2 states that there are 3 32 bit words that make up the UID, but it is ok to read them in different ways. So Making a 64 bit ID is fine.

The 3 words are:

* `0x1FF8 0050` - WAF_NUM + LOT_NUM[55:32]
* `0x1FF8 0054` - LOT_NUM[31:0]
* `0x1FF8 0064` - uniqueIDbits

Now, `hw_get_unique_id()` only uses the latter two words, however **because of the cast `(uint64_t *)` they are read as 64 bit unsigned integers**. this means we're reading 32 bits past the `0x1FF8 0054` value into undefined memory. Since we then shift left 32 bits, we get rid of those undefined bits. However we do the same for `0x1FF8 0064`, which is not shifted. So the higher 32 bits of the UID is a combination of `UID_BASE + 0x04U` and `UID_BASE + 0x018U`. 

I understand fixing this is not easy since there are already devices out there that use this format for UID. So far we have not encountered conflicting UIDs, but I am not sure if uniqueness is guaranteed with the current implementation.

glennergeerts, 2021-05-19:
Thanks for reporting! You are totally right, there is a bug there which results in the UID being wrong... Based on my testing it seems that the UID we generate currently is increased with `0x1100000000000000` compared to the correct one due to this, but it is indeed undefined so I'm not sure if this is always the case.

Fixing it is not difficult, the question is if this breaks anything for users. On our side we are not using the STM32 UID but assign an UID during production from a range we bought at IEEE, which means that we would not be impacted by this. What would happen on your side if this bug is fixed? The UID is stored in the FS and only overwritten (by the output of `hw_get_unique_id()`) when the UID is not set yet in the file. This means that if you would update devices in the field using a fixed implementation of `hw_get_unique_id()` they would still use the same UID. 

As a side note: I'm not entirely sure that uniqueness can be guaranteed, since we are still reducing the 96 bits to 64 bits.

ijager, 2021-05-21:
We do use this implementation of the UID in production. But as you pointed out,  since the UID is cached in the eeprom it shouldn't be a problem to fix it. New devices will just use a different format.

I think the uniqueness will be ok since we're using the `uniqueIDbits` at +0x14.

It is however a breaking change, so maybe it would be a good idea to introduce tags/releases and SemVer version numbers for this stack?

glennergeerts, 2021-05-26:
OK, good to hear it is not really a problem.

I completely agree that we should use semver based releases for the stack. When talking about the network stack I think we can list the following as being part of the public (versioned) "API" to be guarded by semver:
- D7A frame structure (over the air) + D7A protocol (behavior)
- ALP protocol
- d7ap_stack.h and alp_layer.h
- serial modem interface
- D7A system files, OSS-7 specific files and their structure
- embedded LoRaWAN stack updates (MAC version, regional params, ...)

For each of these parts a change can be a patch or minor change (which are both backwards compatible) or a major change. For instance, using a RFU bit in D7A frame structure for a new feature, or adding an ALP operation type or extending a D7A system file with an extra field can all be considered minor changes since they are backwards compatible. To come back to the specific issue with the way the UID is generated: I don't think this is a breaking change; everything will keep on working as discussed above, the file structure remains the same, only the way the UID generation on STM32L is implemented is changed. So IMO this is only a patch release.

glennergeerts, 2021-05-26:
btw the issue should now be fixed in 2666efca3f8a5bd7ec0200179e52adc6a90ed8f9, so I'm closing this one. If you still have a remark feel free to re-open.

commit 39fddc59 (2019-09-27): Merge branch '106-frequency-deviation-and-bitrate-realtime-changable' into 'master'

Resolve "frequency deviation and bitrate realtime changable"

Closes #106

See merge request aloxy/oss-7!75
closed 2021-05-26 by glennergeerts