Sub-IoT/Sub-IoT-Stack #114: Board resets when sending message using default access class at index 1 set to normal rate.

Dond96, 2021-12-07:
I ran into a bug where the B-L072Z-LRWAN1 board would reset when sending a message using access class 1 to normal rate. the code below would cause the board to reset.

```C
dae_access_profile_t profile;
d7ap_fs_read_access_class(0, &profile);
profile.channel_header = (channel_header_t)
{
  .ch_class = PHY_CLASS_NORMAL_RATE,
  .ch_coding = PHY_CODING_FEC_PN9,
  .ch_freq_band = PHY_BAND_868
};
d7ap_fs_write_access_class(0, &profile);

d7ap_send(args); 
```

LOorts-Aloxy, 2021-12-07:
do you have any logs of this? did the board assert somewhere?

Dond96, 2021-12-07:
Nope didn't get an assert the board would just reset and print the logs from the bootstrap function again.

LOorts-Aloxy, 2021-12-07:
hmmm ok, normal rate is automatically being tested, but these tests are very basic. We're not testing low power listening (which is default for access class 0x11) combined with normal rate at this time. Could be that there's still an issue there.

LOorts-Aloxy, 2021-12-07:
and what is the reboot reason when the device reboots?

Dond96, 2021-12-07:
It didn't show a reason it just rebooted. But I just switched to lo-rate for the background message now because I don't need to increased data speed for that.

LOorts-Aloxy, 2021-12-07:
normally one of the first messages when booting is "rebooted with reason X" which should show you the reboot reason. but if it's fixed like this for now, that's also great :)

commit 2c8582fc (2019-11-19): Merge branch '114-lora-duration-and-rssi-is-wrong' into 'master'

Resolve "Lora duration and rssi is wrong"

Closes #114

See merge request aloxy/oss-7!85