Sub-IoT/Sub-IoT-Stack #115: Scheduler hangs when scheduling a task through interrupt service routine

mzaja-be, 2021-12-09:
**The issue**: Task scheduling is not thread safe when attempting to schedule them from the interrupt service routine (ISR). If the ISR tries to schedule a task at the same time as the main thread, the device gets stuck in an infinite loop.

**Probable cause**: The main routine starts scheduling a task. During that period, an interrupt occurs and the ISR also tries to schedule a task. The ISR cannot schedule a task because the scheduler is shown as busy and waits in a loop for the resource to become available. However, since the ISR has higher priority than the main thread, the execution never switches back to the main thread to finish task scheduling, hence the ISR hangs in a loop indefinitely.

The image below shows where it gets stuck:
[image]

**Hardware used**: **B-L072Z-LRWAN1**. Connect PA0 and PA8 pins together.

**Code to reproduce**:
```c
#include "hwgpio.h"
#include "stm32_common_gpio.h"
#include "scheduler.h"
#include "timer.h"
#include "log.h"
#include "debug.h"
#include "framework_defs.h"
#include "platform.h"

#ifdef FRAMEWORK_USE_POWER_PROFILE
	#error "This example can't be used in combination with use_power_profile as the filesystem is not initialized here"
#endif

#define TASK_DELAY ((timer_tick_t)(0.5 * TIMER_TICKS_PER_SEC))

const pin_id_t PIN_INT = PIN(GPIO_PORTA, 0);
const pin_id_t PIN_OUT = PIN(GPIO_PORTA, 8);

// function prototypes
static void drive_output_high_task ();
static void drive_output_low_task ();

static inline void drive_output_high () {hw_gpio_set(PIN_OUT);}
static inline void drive_output_low ()	{hw_gpio_clr(PIN_OUT);}

static void drive_output_high_task ()
{
	drive_output_high();
	log_print_string("Output HIGH");
	timer_post_task_delay(&drive_output_low_task, TASK_DELAY);
}

static void drive_output_low_task ()
{
	drive_output_low();
	log_print_string("Output LOW");
	timer_post_task_delay(&drive_output_high_task, TASK_DELAY);
}

static void drive_output_low_ISR (void *arg)
{
	timer_post_task_delay(&drive_output_low_task, TASK_DELAY);
}

static void init_pins ()
{ 
	// setup output pin to simulate external interrupt source
	hw_gpio_configure_pin(PIN_OUT, true, GPIO_MODE_OUTPUT_PP, true);
	hw_gpio_clr(PIN_OUT);

	// configure interrupt pin
	hw_gpio_configure_pin(PIN_INT, true, GPIO_MODE_IT_RISING, false);
	hw_gpio_configure_interrupt(PIN_INT, GPIO_RISING_EDGE, &drive_output_low_ISR, NULL);
	hw_gpio_enable_interrupt(PIN_INT);
}

// ----------------------------------------------------------------------------
// MAIN
// ----------------------------------------------------------------------------

void bootstrap()
{
	log_print_string("Device booted at time: %d\n", timer_get_counter_value());

	init_pins();

	sched_register_task(&drive_output_high_task);
	sched_register_task(&drive_output_low_task);

	timer_post_task_delay(&drive_output_high_task, TASK_DELAY);
}

```

mzaja-be, 2021-12-09:
The code above should alternate the output of PA8 every half a second. When the pin goes high, ISR is triggered on the rising edge and **drive_output_low_task()** should be scheduled. However, the MCU gets stuck in an infinite loop when attempting to schedule a task from within ISR.

nielsheirman, 2021-12-14:
The issue seems to be caused by the `gpio_int_callback` in`stm32_common_gpio.c` starting an atomic section (and thus disabling all interrupts).

The complete flow causing the bug would be as follows:

1. `drive_output_high_task` is scheduled on the timer
2. the hw timer is configured so it will fire after `0.5 * TIMER_TICKS_PER_SEC`
3. The scheduler will then schedule `__feed_watchdog_task` on the timer (which will stay in the timer queue)
4. Timer fires after `0.5 * TIMER_TICKS_PER_SEC`
5. `drive_output_high_task` is scheduled in the scheduler to be executed as the next task
6. the hw timer is configured to fire for the next task of the timer queue, which at this point is the __feed_watchdog_task. Doing this will set `cmp_reg_write_pending` to true. This value is set to false again by the timer interrupt (thus indicating the hw timer cmp reg is set properly)
7. `drive_output_high_task` is executed by the scheduler
8. `drive_output_high_task` will set the gpio pin high
9. The GPIO interrupt fires
10. `gpio_int_callback` will start an atomic section (disabling all interrupts!) and calls `drive_output_low_ISR`
11. `drive_output_low_ISR` will schedule `drive_output_low_task` on the timer
12. As `drive_output_low_ISR` would need to be executed before __feed_watchdog_task, the hwtimer is configured
13. as the timer interrupt has not yet fired at this point to indicate the cmp reg was set successfully, the processor will hang on `while(cmp_reg_write_pending)`
14. As we are in an atomic section, interrupts are disabled and the timer interrupt will never fire

commit 3cce116b (2019-11-21): Merge branch '115-make-test-suite-great-again' into 'master'

Resolve "Make test-suite great again"

Closes #115

See merge request aloxy/oss-7!86
closed 2021-12-15 by nielsheirman