Sub-IoT/Sub-IoT-Stack #116: Unkown hardfault when running C++ on B-L072Z-LRWAN1

Dond96, 2021-12-09:
This is an issue I ran into earlier but haven't reported yet I just continued in C. 
When trying to build an application with the stack using C++ for the B-L072Z-LRWAN1 it crashes before calling the bootstrap function. I had to figure this out using a debugger. The debugger doesn't give much information about the crash. The code does compile it just crashes runtime. I've also added a picture of the callstack the debugger shows and the local variables of the __hardfault_handler function of stm32_common_system.c where the reason of the crash is "<optimized out>".

```C++
extern "C"
{
  #include "d7ap.h"
  #include "d7ap_fs.h"
  #include "log.h"

  void bootstrap()
  {
    d7ap_fs_init();
    d7ap_init();
    log_print_string("Test");
  }
}
```
[image]

stack in the __hardfault_handler function
[image]

glennergeerts, 2021-12-09:
I have never tried running a C++ application on the stack for embedded, but I would expect name mangling difference between C and C++ being the issue here. I saw that you added an `extern "C"` block around the code, but I wonder if the bootstrap function itself would need to be declared with `extern "C"` as wel? A long time ago we where able to run the stack as part of an OMNET++ based simulation, which meant it was useable from within a C++ program, but this was on the PC host not on the embedded target. You can still see the `__LINK_C` define from `link_c.h` being used in function definitions. My guess is that it would need to be added for `bootstrap()` and probably others as well. If you comment to call to `bootstrap()` do you still crash? Can you break in `__framework_bootstrap()` ?

Dond96, 2021-12-13:
I looked a bit further into the problem and the code seems to crash on assembly code. The code crashed in the init_clock() function of stm32_common_mcu.c file when initialising the RCC structs. The debugger would go into memset assembly code where the program would throw a hardfault. The structs can be seen in the image below. I did put the = {0) in comments so the code would continue.
[image]

Later the code would crash in stm32l0xx_hal_rcc.h line 1201 that line of code would also lead to an assembly file because of calculation with unsigned 32 bit integers. It would lead to __aeabi_dadd.csdasm and would crash on the line of code which can be seen in the figure below.
[image]

So when compiling C++ for the B-L072Z-LRWAN1 the code does crash before the main loop is even called.