Sub-IoT/Sub-IoT-Stack #120: Assertion fail in timer

ijager, 2021-12-21:
I just updated a project to the latest version of the stack (commit dc6307c150744) and now I keep hitting assertion fails in [`timer_fired()` at line 405](https://github.com/Sub-IoT/Sub-IoT-Stack/blob/dc6307c150744b8449f9243c7a9992e25b95dfe7/stack/framework/components/timer/timer.c#L405)

```c 
assert(NG(timers)[NG(next_event)].f != 0x0);
```

Before it crashes it seems to hang (the code seems to loop here) for quite a long time in `void HAL_PWR_EnterSTOPMode(uint32_t Regulator, uint8_t STOPEntry)
`, specifically on the last [line](https://github.com/Sub-IoT/Sub-IoT-Stack/blob/dc6307c150744b8449f9243c7a9992e25b95dfe7/stack/framework/hal/chips/stm32l0xx/Drivers/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr.c#L577):

```c
  CLEAR_BIT(SCB->SCR, SCB_SCR_SLEEPDEEP_Msk);
```

Some debug output before it crashes:

```
Program: start
measurement took 287 ms
[001] sleep (mode 1) @ 2129
[002] EXTI->PR 0
[003] wake up @ 2319
measurement took 288 ms
[004] sleep (mode 1) @ 2623
[005] EXTI->PR 0
[006] wake up @ 6960
[007] sleep (mode 1) @ 10352
[008] EXTI->PR 0
[009] wake up @ 41134
[010] sleep (mode 1) @ 41141
[011] EXTI->PR 0
```

Some notes:

- This is a project with a relatively fast repeating measurement task @ 2Hz. Which is must faster than once per hour we did in another project. Not sure if that could be a trigger for this timer assertion.
- If I do not disable the Watchdog, it keeps working for much longer (a few minutes), however in the end it still crashes at the same place

I am not really sure yet how to debug this further, any suggestions on what steps to take, or what debug output could be useful?

LOorts-Aloxy, 2021-12-22:
Hi! Is this still the case with other sleep modes (changing FRAMEWORK_SCHEDULER_LP_MODE to 0, 1 or 255)? Can you also reproduce it in an independent, bare-bones application?

LOorts-Aloxy, 2021-12-22:
I created a pull request with a possible fix, tried it on the example, and it seems to work, could you test if it's also working for your actual application?

LOorts-Aloxy, 2021-12-22:
that's great news! Happy to help!

closed 2021-12-22 by LOorts-Aloxy