Sub-IoT/Sub-IoT-Stack #132: One-Hop Routing

robotoshi, 2024-03-15:
Hi Sub-IoT community,

I've been looking at and trying Sub-IoT for some time now, and I've just gotten to the point where we need to switch on the one-hop routing that made us choose D7A as our stack in the first place.
However, to my dismay it looks like there is a TODO from 7 years ago saying "TODO hopping not yet supported"!
The website also mentions the one-hop routing, but is it true that one-hop is just completely unsupported?  Is there a plan to implement this any time soon?
I see mention about the ability for another routing protocol to be added, but is this reasonably feasible?

robotoshi, 2024-03-15:
Update: I just found the gitter room where there has been more discussion and it looks like one-hop is not supported.  Can the website be updated to indicate that?  It would have saved me a lot of time to know not to use Sub-IoT in the first place.

glennergeerts, 2024-03-18:
Thanks for bringing this up. I see that one hop topology is mentioned on 
the DASH7 intro page (which explains the spec, not the Sub-IoT 
implementation), or did you see it on another place as well? You are 
right that it would be better to indicate on the website that not all 
spec features are implemented. Will add that. Furthermore, development 
is pretty much on hold (at least from our side), so will add this 
message as well.

--------------lHAmymsvXYHkxb65v6ZcAiM0
Content-Type: text/html; charset=UTF-8
Content-Transfer-Encoding: 8bit

<!DOCTYPE html><html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
  </head>
  <body>
    <p>Thanks for bringing this up. I see that one hop topology is
      mentioned on the DASH7 intro page (which explains the spec, not
      the Sub-IoT implementation), or did you see it on another place as
      well? You are right that it would be better to indicate on the
      website that not all spec features are implemented. Will add that.
      Furthermore, development is pretty much on hold (at least from our
      side), so will add this message as well.</p>
    <p><br>
    </p>
    <div class="moz-cite-prefix">On 3/15/24 01:51, Toshi Taperek wrote:<br>
    </div>
    <blockquote type="cite" ***@***.***">
      
      <p dir="auto">Update: I just found the gitter room where there has
        been more discussion and it looks like one-hop is not supported.
        Can the website be updated to indicate that? It would have saved
        me a lot of time to know not to use Sub-IoT in the first place.</p>
      <p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
        thread.[image]<span style="color: transparent; font-size: 0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;Sub-IoT/Sub-IoT-Stack/issues/132/1998709116</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/Sub-IoT/Sub-IoT-Stack/issues/132#issuecomment-1998709116",
"url": "https://github.com/Sub-IoT/Sub-IoT-Stack/issues/132#issuecomment-1998709116",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------lHAmymsvXYHkxb65v6ZcAiM0--